Kinovarobotics/Kinova-kortex2_Gen3_G3L #216: Inquiry Regarding Firmware 2.6.0 and API 2.7.0 Updates

xyx17, 2025-03-16:
I would like to know the differences between Gen3 firmware version 2.6.0 ([link](https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/gen3/2.6.0/Gen3-2.6.0.swu)) and 2.5.2, as well as between API version 2.7.0 ([link](https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.7.0/kortex_api-2.7.0.post5-py3-none-any.whl)) and 2.6.0. Please also advise if upgrading to these latest versions is recommended. Thank you!

martinleroux, 2025-03-17:
Hi @xyx17 ,

SWU version 2.5 along with the API package v2.6 contained references and functionality for the alpha release of a new configuration with brakes. The project has been abandoned, turns out there isn't enough space left in Gen3 actuators to add brakes that perform adequately, so the latest versions of the firmware and the API simply removed all references to this configuration. We also took the opportunity to clean up the API of fields that were unused by the firmware.

For future updates, you can always refer to our release notes, which are available on [our Artifactory server](https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-documentation-public/Documentation/Gen3/Technical%20documentation/Release%20notes).

xyx17, 2025-03-18:
Hi @martinleroux ,
Thank you for providing the information.
I just want to confirm if I understand correctly—using SWU version 2.6 and API version 2.7 should work without issues and is the recommended approach, is that right?

martinleroux, 2025-03-18:
Correct. You may have some minor errors reported with older code with the API if you were using "unused" fields, but if you are getting any, you can remove these lines and rest assured that the behaviour of your program will not change.