Kinovarobotics/Kinova-kortex2_Gen3_G3L #219: Impedance controller

Reichenbar, 2025-04-21:
## Summary

Is there any impedance controller available for our Gen3 robot?

## Use case

Since we want to achieve some contact-rich manipulation tasks, we hope the robot will be "soft" enough to ensure safety. We know the robot has an emergency stop, but it will interrupt the task execution.

martinleroux, 2025-04-22:
Hi @Reichenbar ,

There is not, but the Kortex API provides all of the necessary tools to implement it yourself.

Reichenbar, 2025-04-23:
Hi @martinleroux , thanks for your response!

I'm not very familiar with robotics control. Could you please point out which specific APIs are most relevant or helpful for implementing impedance control? If there are any example codes or documentation, that would be really helpful too.

Any guidance or reference materials would be greatly appreciated!

martinleroux, 2025-04-23:
You can consider looking at Emprise Lab's public implementation, which is available [here](https://github.com/empriselab/gen3_compliant_controllers). I will say however that if you are a beginner in robotics control, taking on impedance control from the get go will be a challenging endeavour.