Kinovarobotics/Kinova-kortex2_Gen3_G3L #225: Joint angles returned from BaseCyclic API are outside of joint limits

ToruOwO, 2025-06-20:
## Description

When querying the current joint angles, I get out-of-limit values like
[17.721736907958984, 66.60884857177734, 299.3712158203125, 235.97201538085938, 292.9685363769531, 280.77398681640625, 296.1153869628906]

This is for a Kinova Gen3 arm.

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI : latest C++

Kortex Device : Ubuntu 22

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example

```
std::vector<float> KinovaArm::get_qpos() const {
    auto base_feedback = base_cyclic_->RefreshFeedback();
    std::vector<float> joint_angles;
    for (size_t i = 0; i < base_->GetActuatorCount().count(); i++) {
        joint_angles.push_back(base_feedback.actuators(i).position());
    }
    return joint_angles;
}
```

## Expected behavior

Joint angle values in range

aalmrad, 2025-06-25:
@ToruOwO, can you please specifty which API version you are using? If not already used, I advise you to try again with the most recent one (2.7.0).

ToruOwO, 2025-06-28:
Thanks - I updated to 2.7.0 and the issue seems resolved! (was using 2.6)

closed 2025-06-28 by ToruOwO